cpainchaud/RFLink32 #31: Plugin 049 send strange values

franki29, 2021-04-10:
Hi, first of all great software.

I have a wind sensor and the new Plugin 049 now receive the messages, but the output is not correct:

20;AA;LaCrosse-TX141THBv2;ID=1d5fd0;TEMP=fffffe61;HUM=52;BAT=OK;

Currently I am using OpenMQTT to receive the Signal and the values that I get back are OK:
{"model":"LaCrosse-TX141W","id":382461,"channel":0,"battery_ok":1,"temperature_C":9.8,"humidity":81,"test":0,"mic":"CRC","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor","rssi":-68,"duration":696436}

Br. Frank

cpainchaud, 2021-04-15:
Hi!

 I just implemented it a few days ago and I was needing more people to test it! It's working well at home but may be your model is slightly different!

Where do you get this output from ?
{"model":"LaCrosse-TX141W","id":382461,"channel":0,"battery_ok":1,"temperature_C":9.8,"humidity":81,"test":0,"mic":"CRC","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor","rssi":-68,"duration":696436}

cpainchaud, 2021-04-15:
Also please provide the output after you have activated debug ( 10;RFDEBUG=on;  )

I am looking for Pulses right before it prints the output:
**20;XX;DEBUG;Pulses= ............**
20;AA;LaCrosse-TX141THBv2;ID=1d5fd0;TEMP=fffffe61;HUM=52;BAT=OK;

franki29, 2021-04-15:
Thanks for answer. But how can I activate or better reach the CLI interface to set the debug to on?
Currently I am connected through the ser2net interface and get:
20;DB;LaCrosse-TX141THBv2;ID=1d5fd0;TEMP=fffffe69;HUM=2b;BAT=OK;

cpainchaud, 2021-04-15:
Open a telnet connection and send the command there

cpainchaud, 2021-04-15:
Where are you getting this data from?
{"model":"LaCrosse-TX141W","id":382461,"channel":0,"battery_ok":1,"temperature_C":9.8,"humidity":81,"test":0,"mic":"CRC","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor","rssi":-68,"duration":696436}

franki29, 2021-04-16:
HI, debug on causes a crash on my device:
------------- CUT HERE FOR EXCEPTION DECODER ---------------␊
[08:03:18:949] ␍␊
[08:03:18:949] Exception (3):␍␊
[08:03:18:949] epc1=0x40216a5f epc2=0x00000000 epc3=0x00000000 excvaddr=0x4029ad98 depc=0x00000000␍␊
[08:03:18:949] ␍␊
[08:03:18:949] >>>stack>>>␍␊
[08:03:18:949] ␍␊
[08:03:18:949] ctx: cont␍␊
[08:03:18:949] sp: 3ffffd60 end: 3fffffc0 offset: 0190␍␊
[08:03:18:959] 3ffffef0:  00383600 00000001 8000002c 40216aee  ␍␊
[08:03:18:959] 3fffff00:  00000023 00000001 3ffe8594 4020f01d  ␍␊
[08:03:18:968] 3fffff10:  3fff2f40 00000000 401004da b4fdf3b6  ␍␊
[08:03:18:968] 3fffff20:  00000000 4bc6a7f0 000047dc 3fff35c0  ␍␊
[08:03:18:976] 3fffff30:  3ffe8594 3fff12d0 3fff12cc 00000000  ␍␊
[08:03:18:983] 3fffff40:  00000000 3fff1495 00000000 40215565  ␍␊
[08:03:18:983] 3fffff50:  00000000 4029ad7c 3fff14ac 3fff35c0  ␍␊
[08:03:18:990] 3fffff60:  3fffdad0 00004803 3fff148c 4020d0da  ␍␊
[08:03:18:990] 3fffff70:  3fffdad0 00000000 3fff3580 40216a14  ␍␊
[08:03:18:998] 3fffff80:  00000000 00000000 00000001 40100340  ␍␊
[08:03:19:005] 3fffff90:  3fffdad0 00000000 3fff3580 40217528  ␍␊
[08:03:19:005] 3fffffa0:  feefeffe feefeffe feefeffe 40229ba0  ␍␊
[08:03:19:014] 3fffffb0:  feefeffe feefeffe 3ffe8818 4010116d  ␍␊
[08:03:19:014] <<<stack<<<␍␊

franki29, 2021-04-16:
Detail information about the code of my device that is used at OpenMQTT (RTL433) can be find here :
https://github.com/NorthernMan54/rtl_433_ESP/blob/master/rtl_433/devices/lacrosse_tx141x.c

Maybe this will help?

franki29, 2021-04-16:
Using a ESP32 I could successful  enable the debug mode:
20;XX;DEBUG;Pulses=1199;Pulses(uSec)=73,261,832,646,907,563,386,347,384,348,344,383,385,350,591,130,630,102,370,357,619,127,587,129,611,126,369,358,601,129,356,378,625,102,354,382,596,130,614,117,601,130,583,159,568,151,586,145,570,159,336,407,585,137,296,434,324,410,333,392,342,393,305,421,336,402,334,391,578,152,340,392,338,400,569,151,319,418,326,400,319,420,310,413,562,171,534,196,574,157,571,160,562,163,333,407,314,414,313,413,322,406,326,420,556,160,314,415,322,415,553,175,547,181,550,187,529,202,313,412,570,163,304,429,306,428,553,173,555,175,547,181,551,194,290,461,799,668,775,684,788,674,770,697,289,442,298,438,297,428,303,428,540,195,534,200,281,435,536,190,526,202,539,199,283,443,531,201,284,444,528,204,288,442,535,191,526,209,535,194,531,197,537,197,533,212,519,201,275,450,514,220,282,451,280,444,264,471,275,456,276,447,291,445,283,444,527,211,276,454,281,448,524,203,281,460,275,451,261,465,277,452,529,203,524,208,501,245,483,247,503,211,278,449,285,451,276,460,265,457,277,456,520,211,281,459,263,455,525,206,498,235,522,207,515,216,264,466,519,211,278,454,275,457,518,207,517,228,508,215,508,215,271,492,754,700,758,713,736,720,765,694,279,460,262,470,259,466,276,463,502,225,512,216,266,466,512,230,500,218,508,226,269,460,513,215,267,466,511,218,265,465,513,219,510,222,501,228,502,235,500,225,501,219,499,236,259,481,491,228,239,493,261,469,265,465,258,471,258,476,259,471,257,471,508,225,252,476,257,481,498,226,261,469,255,470,261,481,259,464,504,222,495,239,504,231,502,224,504,231,258,472,254,472,260,478,255,474,254,482,484,239,257,471,254,478,502,228,509,223,500,231,492,237,251,483,496,230,253,477,256,480,493,233,500,231,496,236,500,240,240,517,742,708,735,732,751,713,738,721,261,476,252,478,251,480,252,479,492,235,498,246,242,468,496,234,485,246,497,239,244,484,492,240,247,484,486,243,247,484,492,238,491,240,486,245,492,237,493,233,496,250,481,237,242,489,475,258,240,489,245,485,234,499,244,485,243,487,245,483,252,482,492,239,243,483,258,475,496,230,254,487,245,480,251,506,192,538,462,244,486,242,487,243,490,246,485,238,245,488,244,488,249,485,246,473,252,487,493,234,247,495,229,490,494,238,474,254,495,239,489,243,239,488,489,241,249,488,241,485,487,240,493,246,487,235,490,242,245,523,735,728,733,727,720,747,738,723,242,493,246,483,243,484,247,489,487,242,489,241,244,486,487,258,472,245,484,248,240,492,480,247,244,485,493,240,242,488,485,246,492,236,491,241,481,253,482,247,478,241,483,248,243,502,472,247,223,506,244,488,242,489,231,496,240,490,244,485,247,482,494,241,242,489,239,495,481,250,237,492,234,491,244,499,229,495,485,244,468,264,481,244,483,246,484,250,239,492,243,487,234,500,237,492,236,499,476,249,238,495,234,491,485,250,483,247,477,251,476,256,245,482,486,243,246,486,248,489,479,246,479,252,486,250,476,259,227,532,718,733,720,748,731,727,726,737,242,493,240,492,234,495,239,491,482,246,483,255,227,492,478,248,474,260,483,252,235,488,486,242,239,492,482,252,234,496,479,256,477,253,481,247,485,249,476,252,473,264,473,252,232,495,469,260,234,505,232,495,221,507,235,493,239,513,192,556,169,536,467,255,233,493,234,496,485,247,240,501,225,491,230,501,244,486,486,242,487,255,476,247,477,253,485,249,235,497,231,497,226,511,228,496,233,494,484,251,238,499,231,488,490,243,470,261,479,251,481,253,230,497,479,253,233,498,239,492,477,252,474,264,473,250,477,254,473,292,716,732,733,736,717,747,721,741,240,495,236,495,230,497,237,500,474,250,481,249,233,497,483,258,468,253,482,252,231,497,474,256,231,503,476,252,233,498,476,254,478,252,477,252,476,261,479,245,476,254,477,256,235,505,464,255,219,519,225,502,229,502,227,499,231,503,230,502,230,494,485,249,231,499,234,499,477,247,242,493,230,498,234,508,222,497,475,255,473,260,477,253,471,260,477,256,228,500,229,500,231,498,236,494,237,497,469,263,224,503,233,502,472,262,469,255,476,253,469,263,231,502,472,255,227,502,235,498,477,248,479,255,478,257,470,267,222,1981,707,750,724,735,725,737,728,739,239,488,242,493,237,496,230,511,463,259,474,254,230,498,480,256,474,250,476,256,232,499,483,250,234,492,480,252,234,498,482,248,475,256,480,245,469,270,463,253,477,259,442,326,164,562,440,259,225,502,231,495,234,497,238,495,234,497,232,501,477,251,234,497,231,498,233,509,224,497,217,510,232,502,234,497,232,500,226,497,236,500,476,247,481,254,478,256,230,502,228,497,229,502,241,488,235,503,228,493,220,515,235,493,234,494,238,493,240,490,240,491,479,255,480,251,237,491,481,256,477,248,476,252,232,497,235,504,232,531,712,736,724,739,729,737,724,742,235,491,240,502,232,492,235,496,477,256,478,259,227,494,483,245,474,255,486,244,242,494,477,252,232,500,481,253,228,498,477,251,483,260,468,252,476,254,468,266,472,263,469,252,220,514,473,255,235,495,232,503,228,497,234,502,228,501,234,500,471,253,236,509,220,500,223,502,237,491,241,495,237,489,229,503,240,494,236,493,232,497,485,248,475,257,475,251,242;RSSI=-9999;REASON=Unknown␍␊
[08:15:19:124] 20;03;LaCrosse-TX141THBv2;ID=1d5fd0;TEMP=fffffe2b;HUM=4f;BAT=OK;␍␊

cpainchaud, 2021-04-16:
Yes thank you I have what I need for now! :)

cpainchaud, 2021-04-16:
Here is what I get after a fix:
20;01;LaCrosse-TX141THBv2;ID=5d5fd;CHN=0000;TEMP=002b;HUM=4f;BAT=OK;

Remember this is RFLink protocol so most values are hexadecimal and can be translated to:
20;01;LaCrosse-TX141THBv2;ID=382461;CHN=0000;TEMP=4.3°C;HUM=79%;BAT=OK;

cpainchaud, 2021-04-16:
You can test last firmware directly from the UI:
[image]

franki29, 2021-04-16:
Yes, that's looks much better now, only the humidity value seems to be wrong:
While OpenMQTT shows 55%, RFLink is showing 37%. (A other sensor with BME280 shows 62%).
The temerature values are the same.

franki29, 2021-04-16:
Also the values for the wind speed is different. 
Additional I get a message 20;7D;PONG;
What does that mean?

cpainchaud, 2021-04-16:
> Also the values for the wind speed is different.
> Additional I get a message 20;7D;PONG;
> What does that mean?

if you plugged Ser2Net to Domoticz/HomeAssistant, they send a PING command from time to time to check that the device is still online. RFLink response is broadcasted to all people connected.

cpainchaud, 2021-04-16:
> Also the values for the wind speed is different.
> Additional I get a message 20;7D;PONG;
> What does that mean?

For Windspeed I need you to capture a packet which shows windspeed (like you did for temperature) so i can look at it

franki29, 2021-04-16:
This sould be the one for windspeed 
20;XX;DEBUG;Pulses=456;Pulses(uSec)=237,498,237,497,234,496,239,497,231,495,240,493,243,492,238,492,240,496,234,495,237,494,245,491,482,246,242,496,482,247,481,252,485,250,235,495,243,490,484,247,239,534,735,733,732,732,735,728,736,729,246,492,241,495,235,492,243,493,483,252,482,246,242,491,484,249,479,254,482,252,235,494,483,254,231,495,484,252,243,493,480,248,483,250,483,250,480,252,484,253,478,249,481,257,230,498,481,253,234,495,238,497,240,495,232,496,239,495,234,496,486,250,237,495,237,494,237,497,235,494,242,493,238,493,240,492,241,493,485,244,482,254,484,249,487,243,483,252,237,493,240,496,233,502,234,492,241,492,237,493,245,492,235,496,239,495,240,493,240,488,244,492,486,240,244,493,482,252,483,249,480,253,236,495,243,490,486,251,235,532,736,729,734,730,737,732,728,734,247,490,237,500,237,490,242,491,485,245,488,248,238,497,481,250,478,256,479,252,237,495,481,248,236,501,477,257,236,494,477,256,480,254,480,251,479,253,480,253,479,255,477,258,230,497,482,252,236,495,237,498,230,502,233,497,237,498,232,497,482,252,236,499,234,494,241,495,235,493,240,495,238,497,232,497,238,497,481,247,488,248,479,253,481,249,481,253,237,495,240,492,240,495,236,496,237,495,236,496,238,497,236,495,235,494,240,494,238,499,237,489,486,250,236,494,486,246,482,254,483,246,242,494,235,493,485,251,238,536,730,734,733,734,725,737,730,739,242,492,237,498,236,495,237,496,481,250,479,253,233,501,482,248,480,254,485,244,241,495,480,252,235,495,481,255,235,494,486,249,477,255,483,249,479,255,480,255,480,249,480,255,233,498,476,256,237,499,232,496,239,495,234,495,239,495,240,494,481,247,243,492,234,502,237,491,242,494,235,493,240,493,241,495,234,496,484,249,483,248,487,246,482,253,482,245,242,493,235,501,235,493,244,491,233,495,241,495,242,492,236,493,243,492,239,490,242,493,240,495,483,245,239,496,483,249,485,249,481,252,237,490,243,495,484,247,239,5000;RSSI=-9999;REASON=Unknown
20;C1;LaCrosse-TX141THBv2;ID=5d5fd;CHN=0000;WINDIR=000;WINSP=001f;BAT=OK;

cpainchaud, 2021-04-16:
ok thank you, what should be the correct speed? for my information ?

franki29, 2021-04-16:
RFLink : 20;08;LaCrosse-TX141THBv2;ID=5d5fd;CHN=0000;WINDIR=000;WINSP=0011;BAT=OK;
OpenMQTT : model":"LaCrosse-TX141W","id":382461,"channel":0,"battery_ok":1,"wind_avg_km_h":1.7,"

cpainchaud, 2021-04-16:
please try the the latest firmware and let me know

Wind needs to translated from hexa to decimal then divided by 10:
WINSP=001f where 1f = 31  then speed = 31/10 = 3.1km/h

franki29, 2021-04-16:
Hmm, WINSP=0011 it 11 so DEC 17 so 11 is the correct value.

Means that the wind speed has the correct value, must be something wrong from my side ( I am using note red to send the data to domoticz)

Same with the Humidity, looks like Domoticz is using wrong values as in the mqtt message it is 2d Hex, 45 in decimal.

So looks like from your side everything is OK now
Thanks for helping

cpainchaud, 2021-04-16:
Domoticz has a direct integration with RFLink, did you try it first? Then you can tell if Domoticz or NodeRed is the issue

franki29, 2021-04-16:
Domoticz direct integration shows right wind but wrong humidity data.
The problem with node red and openmqtt can be fixed by multiply the values by 3.6. in node red

Currently the direct integration of Domoticz is showing 2% humidity.

cpainchaud, 2021-04-16:
Please allow me to review official RFLink output and Domoticz+HomeAssistant code for humidity setting, may be RFlink source code which we are using as a base is wrong and should be decimal !

cpainchaud, 2021-04-16:
@franki29 I have checked HA, Domoticz and Jeedom and they all expect decimal values so I have made the change for all plugins, it's now live the new& firmware

franki29, 2021-04-16:
Yes, now it is working and and the results are now the same for RFLInk and OpenMQTT

Thanks again for your help

closed 2021-04-16 by franki29