cpainchaud/RFLink32 #40: SX127X how-to ?

LemarinelNet, 2021-06-29:
Hi,

Just seen a post on Arduino Forum about this project !
Really nice work ! 

I'm planning to use this for Somfy RTS blinds, so I need a special frequency...
Instead of finding a weird frequency module, I've read that SX127X is supported ? 

Is there any instructions on how to implement it ? 

I've seen an ESP32 module with an SX1278 and an OLED display on amazon,  just 19€, seems a good option ?

obones, 2021-07-05:
The 433MHz RFM69CW module works fine for RTS and is already supported out of the box.

cpainchaud, 2021-07-19:
> I've seen an ESP32 module with an SX1278 and an OLED display on amazon, just 19€, seems a good option ?

Hi!

I suggest that you don't buy integrated modules with ESP32+radio module : as far as I know, all such cards on the market are not compatible because the PINs used to connected radio modules are either INPUT or OUTPUT so you can use them as to SEND and RECEIVE. You will need to buy an ESP32 and a RFM69HCW or SX1278. @obones mentionned that RFM69HCW might be the easiest to operate and he is right

LemarinelNet, 2021-08-02:
Thanks for the RFM69 option, I'll source one and give it a try.
Is there any wiring guide for RFM69 ?

obones, 2021-08-17:
This is what I have configured in the radio tab:
[image]

And I have the following pin mapping:

RFM69CW | ESP32-WROOM32-D
------------ | -------------
3V3 | 3V3
GND | GND
NSS | 5
SCK | 18
MOSI | 23
MISO | 19
DIO2 | 25
RESET | 4

And of course an antenna plugged on the ANT pin of the RFM69 module. In my case, this is just a 17cm straight wire as I'm targeting the 433MHz band.

LemarinelNet, 2021-08-20:
Thanks for you reply.
I have received a SX1278 module. I've wired it as your RFM69, and setup Hardware to SX1278.
The radio seems to be correctly init'ed according to logs, but it doesn't seems to receive anything: I have some Somfy remotes, and a garage door remote on 433Mhz.
Any button I push on both remotes doesn't trigger anything in serial console ? Is there something to enable so received pulses are shown ? 
Is there anything to do to help debugging with SX1278 ? I seen that it has a DIO0 pin, that I wired as your's DIO2 pin.

Thanks for your help :)

obones, 2021-08-20:
With these modules, you have to make sure they are initialized to 433.42MHz and not the more regular 433.92MHz 
This is visible in the serial log, and can be controlled with the command line interface on the serial link, like this:

    10;config;set;{"radio":{"rx_frequency": 433420000}}

You can also enable the debug mode that shows data frame that are not understood by any plugins with this command:

    10;RFUDEBUG=ON;

Note that on the 433.42MHz, you will only find RTS remotes, note the more recent IO HomeControl ones which work on the 868.95MHz band, use GFSK modulation along with AES encryption (most probably). Those ones are totally out of the RFLink32 scope

LemarinelNet, 2021-08-21:
I have tried with the RFDebug mode enabled, but even with that, I don't see any receiving on the serial console. Strange !? Perhaps a wrong wiring ? 
How is supposed to work the DIO0 Pin on the SX1278 ? The same as the DIO2 of RFM69 ? 

I also tried as you suggest changing the frequency, but I still have complete silence on the console when pushing my somfy remotes :( 

My garage door is not a somfy, neither the remote, and so it should be on the regular 433.92 frequency.

I seriously doubt of my wiring, despite the console saying it is init'ed successfuly :

Initialized SX1278(freq=433.42Mhz,br=9.600kbps,rxbw=250.0khz)=0
SX1278 setOOK=0
SX1278 set encoding result=0
SX1278 set data shaping result=0
SX1278 setOokThresholdType(2)=0
SX1278 setOokFixedThreshold(0x0C)=0
SX1278 setOokPeakThresholdDecrement() result=0
SX1278 disableBitSync() result=0
SX1278 setGain() result=0
Hardware initialization was successful!

I also see a "warning" just before radio logs : 
Start of async Receiver was requested but it's not enabled!

Isn't it a problem ?

LemarinelNet, 2021-08-29:
Some progress !

In the 1_Radio.cpp file, there was line 1148 : `      radio.setFrequency(433920000);`
Which I changed to params::frequency.

Now, I see my somfy remotes emitting.

But when I try to create a "RFLink somfy remote" to be paired with the cover, the ESP sends a Guru Meditation Error : There is a IntegerDivideByZero error; which seems to occurs at RFLink/./Plugins/Plugin_017.c:567 
This line is 
`const int RTS_HalfBitPulseDuration = 640 / RawSignal.Multiply;`
And the following lines are also dividing by RawSignal.Multiply.

I suspect that this variable is not initialized..?

obones, 2021-08-30:
This is fixed by the following PR: https://github.com/cpainchaud/RFLink32/pull/39

LemarinelNet, 2021-08-30:
Ok, I haven't seen this one, and since it wasn't merged, I did not tried it !

Have a look at my last PR, there is a lot of fixes around RTS and Ser2Net...

closed 2021-09-01 by LemarinelNet