cpainchaud/RFLink32 #44: Multiple radio modules at the same time ?

LemarinelNet, 2021-09-01:
Hi,

I see that the original RFLink allows to use also NRF24L01.

I haven't search more to see if it can be used at the same time than the "standard" frequency (433,868....), but...

Would it be possible to implement more than one radio at the same time ?

It would need : 
- Handling the loop on one or many radio modules
- Handling initialization of more than one module (SPI modules would each use one CS/SS/NSS pin)
- Handling multiple Radio pins settings

But it would be a killer feature... 
BTW it would allow to also implement MySensors gateway mode...

cpainchaud, 2021-09-10:
Hi,

 On the paper it would be possible but checking for both devices at the same time will make timings less accurate. Worse, when receiving & processing message through plugins, you would just 'ignore' what happens on the other device.
  There could be 'dirty' hacks potentially like using ESP32's second CPU core (each device would be 'looped' and processed in a different core, there are other catches there as well like the fact that Core 0 can be interrupted anytime by other async tasks (wifi, IOs ... etc)
 There cleanest way I could be thinking of would be to use an additional & cheap microcontroller, like an ATTiny414 which costs like 50cents, it would be in charge of extracting the timings and send them to UART/Serial, ESP32 would do the signal processing portion.
 
 Is it something you might be interesting to help developing ?

LemarinelNet, 2021-09-18:
Well if it has been in my knowledge, I would have helped to do it, but I never used the second core, and it seems to be a huge task.
It would also require to add some kind of locking to avoid that both cores talk at the same time (to serial, or ser2net, etc..)? 
The cheap controller seems a good way , but it would make harder to OTA update the whole thing !

PE1DDA, 2023-12-26:
Hello,

I use the original RFLink for many years now and I made a hardware addition (NE555 + CD4066) that switches between the output of a 433 mHz and a 868 mHz receiver every 60 seconds to the input of the Atmega for reading the information of some 433 thermometers and a 868 fine offset weather station. This works fine for me.
My question for you, is there a simple way to make a second input pin on the ESP32 available for a 868 mHz receiver and a software option to have it switch between 433 / 868 inputs with a definebale switch time? I think many users won't mind the timing, nor the none simultaneous checking of the receivers.