cpainchaud/RFLink32 #45: Branch RTL _433 on ESP32 is craching after some time (Master is working fine)

franki29, 2021-11-21:
Hi all,
I am testing the RTL_433 branch as the master did not detect inFactory-TH sensor.

But after ~5 Minutes the device suddenly reboot and than completely stops working some time later.
I am using Ser2Net connection with Domoticz and MQTT Server.

The Problem starts with MQTT Error showing on the RFLINK website.
There is no crash message at the serial connection.

But Domoticz is crahsing suddently.

Domoticz shows :

021-11-21 11:16:08.113 Status: RFlink: connected to: 192.168.2.27:1900
2021-11-21 11:16:21.073 Status: RFlink: disconnected
2021-11-21 11:16:21.074 Status: RFlink: Connection reset!
2021-11-21 11:16:51.110 Status: RFlink: connected to: 192.168.2.27:1900
2021-11-21 11:17:37.728 Status: RFlink: trying to connect to 192.168.2.27:1900
2021-11-21 11:17:37.807 Status: RFlink: connected to: 192.168.2.27:1900
2021-11-21 11:17:37.728 Error: RFlink: Nothing received for more than 30 seconds, restarting...
2021-11-21 11:17:37.728 Error: ASyncTCP: connect called while socket is still open. !!!
2021-11-21 11:18:17.732 Error: RFlink: Nothing received for more than 30 seconds, restarting...
2021-11-21 11:18:37.733 Status: RFlink: trying to connect to 192.168.2.27:1900
2021-11-21 11:18:37.741 Status: RFlink: connected to: 192.168.2.27:1900
2021-11-21 11:18:37.734 Error: ASyncTCP: connect called while socket is still open. !!!
2021-11-21 11:19:17.738 Error: RFlink: Nothing received for more than 30 seconds, restarting...
2021-11-21 11:19:37.740 Status: RFlink: trying to connect to 192.168.2.27:1900
2021-11-21 11:19:37.927 Status: RFlink: connected to: 192.168.2.27:1900
2021-11-21 11:19:37.927 Status: RFlink: disconnected
2021-11-21 11:19:37.740 Error: ASyncTCP: connect called while socket is still open. !!!
2021-11-21 11:19:37.927 Error: RFlink: Bad address 

When I try to swich back to Master using the web gui I get :

Backtrace: 0x40088980:0x3fff8cb0 0x40088bfd:0x3fff8cd0 0x401a6dab:0x3fff8cf0 0x401a6df2:0x3fff8d10 0x401a671d:0x3fff8d30 0x401a67f4:0x3fff8d50 0x400ee295:0x3fff8d70 0x400ee6f7:0x3fff8d90 0x400eec6d:0x3fff8dc0 0x400ee479:0x3fff8e10 0x400ecdea:0x3fff8e80 0x400ecd4d:0x3fff8ec0 0x400db803:0x3fff8ee0 0x401b0a1b:0x3fff8f50 0x400efa45:0x3fff8f70 0x400ed8e9:0x3fff8fc0 0x400ed9b1:0x3fff9000 0x400edc05:0x3fff9050 0x401a7f65:0x3fff9070 0x401a7fe1:0x3fff90b0 0x401a8686:0x3fff90d0 0x40089c0e:0x3fff9100

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8

Arduino IDE Version :	10805
ESP SDK version:	v3.3.5-1-g85c43024c
Sketch File :		RFLink/RFLink.cpp
Compiled on :		Nov  9 2021 at 08:55:41
20;00;RFLink_ESP;VER=5.1;BUILD=20211109-085345-rtl_433;
Loading persistent filesystem... OK. File system usage: 12/320KB.
Now opening JSON config file '/config.json'
JSON file mem usage: 1791 / 4096

Hope all make sense

Br. Frank

cpainchaud, 2022-01-03:
Hi,

 Your explanations are  a bit confusing sorry.
 What is the error logs you pasted earlier? The Upgrade logs or the Crash logs?

franki29, 2022-01-03:
Hi, sorry for the confusion.

The first Log comes from my Domoticz  installation.
Do not worry about this one.
I just wanted to show, how the behavier lokks like.
So the ESP32 device is running ~ 3 Minutes and than it is stopping communication.

The second one is the crash log when I try to update over the Web Gui of the ESP32 device from RTL_433 back to Matster version.

So I uploaded the master version over USB cable.

Since opening the issue, the same device is now running stable with the master version, so the hardware and power supply seems to be OK.

There is no crash log when I run the RTL Version, it is just stopping the communication and it is not reach able over the WEB-GUI, but can be still ping in the network.

Hope this helps a little bit more.

One thing, the inFactory-TH sensor. is sending ~8 times the same message and I have two of them.
Maybe some kind of buffer overflow? As I am not a expert is is just guessing.

Br. Frank

cpainchaud, 2022-01-03:
It's normal for devices to send their messages a few times, they all  do that for various reasons.

RTL433 version pushes the ESP32 to its limits in term of memory and also it is much much less tested (fewer people is running it) so instability is not surprising at this stage.

Did you enable RFDEBUG mode so we can first check what signals are captured by the Radio? based RTL_433 plugin for your device, our default limit of 5000ms could be too small and may require a few tweaks for your specific protocol. Would be nice to see what your signal looks like, you could even write a native plugin for your device

franki29, 2022-01-03:
Hi, I will enable RFDEBUG and come back with the result, but it will take a while.
I am not a programmer, so I will not be able to write a plugin, sorry.

closed 2023-01-16 by cpainchaud