cpainchaud/RFLink32 #49: CC2500 Support howto question

petervflocke, 2022-02-06:
Hi,

Great idea to port RFlink from Mega to ESP and inluce MQTT, very much appreciated!

Does your code support CC2500 ([RFlink](https://www.rflink.nl/wires.php)) module?
And if it does, how to wire it?

I look forward to hearing from you
Thank you
Peter.

cpainchaud, 2022-02-17:
hmm I don't know of any OOK on 2.4Ghz band :D are you sure that RFlink is what you need? not a zigbee/thread/BLE type project instead ?

petervflocke, 2022-02-17:
RFLINK supports Ikea Ansluta / Utrusta using the 2.4Ghz frequency to control dedicated UTRUSTA end OMLOPP led lights.
Link here: https://www.rflink.nl/devlist.php at the very end of the page. 
And the wiring here: https://www.rflink.nl/wires.php also at the end of the page. 
This is just another communication module and I thought that you can check/integrated this as well.

Now I believe it was my big mistake to perceive your project as a kind of "fork" from the above mentioned RFLINK one.

Anyway thanks for answering.
Cheers
Peter

cpainchaud, 2022-02-17:
no we are a fork but we don't have thousands of users. 

First time I see  these 2.4Ghz remotes, but we learn everyday.

petervflocke, 2022-02-17:
> no we are a fork but we don't have thousands of users.
Could you then think about my "ticket" as a feature request?
Just ping me once you (might) got it working.
Thx.
Peter