cpainchaud/RFLink32 #52: RFM69 Wiring

hallard, 2022-02-28:
Hi There,
I compiled and tried this awesome soft on ESP32 and wired 433MHz RXB6 all is working fine as you can see got data from remote from Sonoff and Generic LED remote
```
20;48;EV1527;ID=030290;SWITCH=01;CMD=ON;
20;49;EV1527;ID=030290;SWITCH=0d;CMD=ON;
20;4A;EV1527;ID=030290;SWITCH=06;CMD=ON;
20;4B;EV1527;ID=030290;SWITCH=06;CMD=ON;
20;4C;EV1527;ID=030290;SWITCH=07;CMD=ON;
20;4D;EV1527;ID=0d0126;SWITCH=01;CMD=ON;
20;4E;EV1527;ID=0d0126;SWITCH=02;CMD=ON;
20;4F;EV1527;ID=0d0126;SWITCH=08;CMD=ON;
20;50;EV1527;ID=0d0126;SWITCH=04;CMD=ON;
```

I also have some boards with RFM69 used for other project and looks like I'm unable to get the same signal that I receive with RXB6.
Both board are running on my desk, I receive on one not on the other. RFM69 looks like seen an init 
```
03:14:12.819 -> RFM69 SetOOK()=0
03:14:12.819 -> RFM69 setDataShapingOOK()=0
03:14:12.857 -> RFM69 setEncoding()=0
03:14:12.857 -> RFM69 setOokThresholdType(2)=0
03:14:12.857 -> RFM69 setOokFixedThreshold(0x06)=0
03:14:12.857 -> RFM69 setOokPeakThresholdDecrement() result=0
03:14:12.857 -> RFM69 setLnaTestBoost()=0
03:14:12.857 -> Hardware initialization was successful!
03:14:12.857 -> Applied slicer 'Legacy'
```

So is there anything to wire for RFM69 more than MISO/MOSI/CLK/CS and power ? BTW do I need to wire any DIO of module or some other pins? 

These boards receive signal send by moteino with RFM69 library (Felix's one) so it's not an HW default, I'm pretty sure I'm missing something stupid. 
Here below
[image]

Thanks for any advice.

cpainchaud, 2022-03-01:
I see you are missing the RX Data pin configuration :)

hallard, 2022-03-01:
Ah ah excited to test, was sure I was missing something, used intensively RFM69 but never tried this mode, will test at home tonight and close if it work :-)
BTW this means that pin (DIO2) could be changed in software? Just asking because I already have some old boards routed so if I can avoid cutting trace and solder wire, worth trying to find where it's configured in the code.
Thanks

cpainchaud, 2022-03-01:
no need for rewriting, just looking at the PIN where it's connected and configure  "RX Data" in the UI

hallard, 2022-03-01:
Sorry my bad, I was talking of the pin used on RFM69 side (the one that out the RF signal) not the one connected to the CPU device that we can change on the UI of course.
Imagine I want that the RF signal out from DIO1 instead of DIO2 (because my boards are routed like that) then I think I need to change that in the driver side of the RFM module. But may be I'm wrong.

cpainchaud, 2022-03-01:
I don't remember if RFM allows you to change the DIO port assignment, some brands do, other don't

hallard, 2022-03-01:
I need to check back documentation but il all case if it's possible it's here [here](https://github.com/jgromes/RadioLib/blob/master/src/modules/RF69/RF69.cpp#L221-L223) from `RadioLib`

Anyway, just wired DIO2 from RFM69 to my D1 GPIO and it works as expected, thanks for the trick.

closed 2022-03-01 by hallard

hallard, 2022-03-01:
Just checked documentation for continuous mode no way to change data pin, it is coming from GPOI2 also RadioLib provide (as datasheet advice) DCLK on GPIO1

in Continuous mode it is always recommended to enable the bit synchronizer to clean the DATA signal even if the
DCLK signal is not used by the uC (bit synchronizer is automatically enabled in Packet mode).

cpainchaud, 2022-03-02:
more importantly now : can you see some signals ? :)