cpainchaud/RFLink32 #58: Support for Heltec Lora 433Mhz

calibrae, 2023-01-02:
Hello

Just received this little board, and I'm trying to make a RFLink bridge out of it. 
It"s ESP32 based so obviously I've been able to flash RFLink32 but no packet is received by the 433 radio. 
This little device being very cheap, already embeding the RF transceiver, and sporting a LiPo connector, it appears to be a good choice. 

@cpainchaud Since you're french, and parisian, if needed, I'd be happy to provide a board for you to add support. My knowledge of embeded programming is way too light for handling this myself as of now. 

thanks for the great job

NB: this is the board I use https://www.aliexpress.com/item/33018609928.html

Cheers

cpainchaud, 2023-01-03:
Hi,

 I do own a very similar board (may be it's the same but you never know with chinese cloning clones :)
 Here is a post I made in Heltec's forum: http://community.heltec.cn/t/wifi-lora-32-v2-design-issue/4058

 Basically you should be able to RECEIVE but not transmit out of the box because the pin they used for DIO2 of the SX127X is not capable to be used as an output. (You could still do some soldering to change the pin if you want).

 All what you would have to do to make it work is to change the platformio.ini 's board to be heltec's rather than generec ESP32.

calibrae, 2023-01-03:
Yes it totally looks the same. 
Too bad for the reception power. 
I'm a total noob with platformio. Is there a readme somewhere I can use? 
Thanks

cpainchaud, 2023-01-03:
https://docs.platformio.org/en/stable/boards/espressif32/heltec_wifi_lora_32_V2.html

board = heltec_wifi_lora_32_V2

cpainchaud, 2023-01-16:
closed until you found an issue ....

closed 2023-01-16 by cpainchaud