cpainchaud/RFLink32 #61: using frequence 868Mhz

geromey, 2023-01-29:
Hello,

I used your RFLink32 for running on shutters 433Mhz and it does work very well.

I would like to use it this time for shutters working on 868Mhz. I have a module which work on 868Mhz but I don't manage to receive anything for 868Mhz remotes. is there a configuration or libraries I should change in order to make it work on 868Mhz ?

Thanks a lot for this great project !

cpainchaud, 2023-01-29:
Hi,

 It's all about your receiver, what radio hardware are you using? Also if you don't mind what are the shutters brand and model?

thank you

geromey, 2023-01-29:
This is Heltec WiFi-LoRa 32 V2. The shutters are Provalux.

Thank you !

cpainchaud, 2023-01-29:
this board has a built in receiver which is tuned for a specific radio band, did you purchase the 868Mhz one?

geromey, 2023-01-29:
Yes

cpainchaud, 2023-01-29:
ok then did you select hardware SX1278 in Radio settings? If so then I just realized the UI is not showing the frequency parameter :D
in serial type this command:  10;config;set;{"radio":{"rx_freq":868950000}}
That will set frequency to 868.95 Mhz

That said can you tell me which hardware you used for your 433mhz receiver? Because Lora Heltec cannot transmit with their current pinout as far as I can remember

cpainchaud, 2023-01-29:
Oops I meant hardware = SX127**6**

satinebm, 2023-01-29:
Argh, the hardware failed to initializ

```
Now trying to initialize hardware 'SX1276
Initialized SX1276(freq=868.95Mhz,br=9.600kbps,rxbw=250.0khz)=-2
SX1276 setOOK=-16
SX1276 set encoding result=0
SX1276 set data shaping result=-107
SX1276 setOokThresholdType(2)=-16
SX1276 setOokFixedThreshold(0x0C)=-16
SX1276 setOokPeakThresholdDecrement() result=-16
SX1276 disableBitSync() result=0
SX1276 setGain() result=-16
Hardware failed to initialize, we will retry later!
```

cpainchaud, 2023-01-29:
are you using the official firmware from downloads page or compiled it yourself? Also @satinebm  is not @geromey so are we talking about the same hardware issue here?

satinebm, 2023-01-29:
Yes we are talkin about the same hardware issue. Sorry for the confusion.
We are using the official firmware.

cpainchaud, 2023-01-29:
ok official precompiled firmware is created for generic ESP32 dev board.

Each board maker is making its own pinout for the various usages. In your case you have 2 solutions:
 - re-compile yourself with the right board ID in platformIO (heltec_wifi_lora_32_V2) and in Radio settings/menus change RX CS, RX Reset, RX data (SX1272's DIO3)
 - in Radio settings/menu : 'guess' what the right pins are for RX CS, RX RESET, RX DATA from Heltec docs
 
I have on such board around and will try to help you tomorrow, may be right a KB about it

geromey, 2023-01-30:
Hello Christophe,

We start to unerstand how settings works. We are not sure which pin is the RX DATA.

Here is the pinout of our board. Could you tell which one it is ?

https://www.passion-radio.fr/img/cms/WIFI_lora_32_pinout_diagram.jpg

caliKev, 2023-03-23:
We’re you able to get this working? I’m trying to do the same. I believe rx data is pin26

cpainchaud, 2023-03-24:
@caliKev first of all you must ensure that you have compiled the firmware for your board. the pre-compiled ones are for ESP32-Dev-board. 'heltec_wifi_lora_32_V2' is the one you should pick

closed 2023-11-08 by cpainchaud