cpainchaud/RFLink32 #63: Changes of the WiFi network makes the Sonoff RF Gateway R2 inaccessible

dilerian, 2023-02-01:
Hi,
first, thank you very much for the continued development of this awesome firmware.
I noticed the following on the Sonoff RF Gateway R2:

1) Configuration - both the STA and the AP mode are active on the device. The device is configured to connect as a client to a WiFi router. The Serial2Net interface is enabled. Everything functions as expected. The device receives an IP address from the router by DHCP and a computer can connect to it both via the received IP address and by a direct connection to the device AP with the IP 192.168.4.1.

2) Issue - if the router is reconfigured or switched off (i.e. the WiFi network, to which the device connects as a client, is not present any more), the computer is able to connect to the device AP but cannot load the WebGUI from it. The Serial2Net interface is also inaccessible.

The computer receives an IP address form the device AP and the browser reaches the prompt for login and password at 192.168.4.1. After that, the GUI becomes unresponsive. Pings to the device at 192.168.4.1 are intermittent - the ones received have a delay of about 500ms. The Serial2net interface is unresponsive.

I assume that the device enters some infinite loop trying to connect to the non existing router and does not have enough resources to process the WebGUI requests any more.

3) UX problem - I could not find any factory reset sequence to apply in this case apart from reflashing the whole firmware by serial adapter.

cpainchaud, 2023-02-03:
Hi,

 I suppose the issue persists even if you reboot the device?

 note: from Serial/Serial2Net you can reset the config with command 10;config;reset;

dilerian, 2023-02-03:
Hi,
yes, the issue persists after a reboot and happens with two different routers. The Serial2Net reset and reboot commands function as expected provided that there is a functioning connection to the device. The problem is that after a change/removal of the WiFi network, any client/user gets locked out without the possibility of a connection or a factory reset without reprogramming the firmware by a serial console or bringing back the WiFi network with the same SSID and WiFi password.