cpainchaud/RFLink32 #74: Plugins 048 & 101 for decoding Oregon Scientific Protocols not implemented??

tkotz8105, 2023-04-03:
I have several sensors using the Oregon V1 and V2 protocols that are not being recognized even with moving the senors close to the rf433 card.  I noticed in the code that the plugins for the Oregon sensors (048 and 0101) are not used.... filename extensions were changed from 'c' to 'old'.  I've looked at the code to see if one of the other plugins perform that function but I couldn't figure it out.  Any explanation & help are greatly appreciated.

tkotz8105, 2023-04-23:
@yndtrud Thanks ... I'll try it

nona00, 2023-06-12:
> [oregon_20230422.zip](https://github.com/cpainchaud/RFLink32/files/11302157/oregon_20230422.zip)

for me the patch works with 
Oregon UV138 UV Sensor (EA7C) and 
Indoor Temp/Hygro: THGN122N, THGN123N, THGR122NX, THGR228N, THGR238, THGR268, THGR122X

would be great if the patch could be merged

cpainchaud, 2023-11-09:
@yndtrud 's patch manually merged.

thx

closed 2023-11-09 by cpainchaud