cpainchaud/RFLink32 #78: CC1101 Not working

EBME2, 2023-05-20:
I seem to get an error continuously

Now trying to initialize hardware 'CC1101'
Initialized CC1101(freq=433.92Mhz,br=9.600kbps,rxbw=250.0khz)=-104
CC1101 SetOOK(true)=0
CC1101 setPromiscuousMode(true)=0
CC1101 disableSyncWordFiltering(true)=0
Hardware failed to initialize, we will retry later!

I have tried two CC1101 modules both the same, I can however get the CC1101 module to talk to the ESP32 using this project which is using the same SPI interface pins, and I have configured the pins in the web interface.

https://github.com/NorthernMan54/rtl_433_ESP/tree/master

Has anyone managed to get a CC1101 working?

EBME2, 2023-05-23:
I think I have figured out why I get the above error, the CC1101 does not like a 'rxbw' of 250khz and is returning an error -104. From what I can see the Receive Bandwidth setting should be the radio section of the UI but isn't for me. I can manually set the rxbw in the 1_Radio.cpp and the CC1101 now initializes at least.

Still don't think things are working but at least a step forward.

cpainchaud, 2023-05-23:
Hi,

 You should be able to change RX bandwidth with (CLI/Telnet only) command:
`10;config;set;{"radio":{"rx_bandwidth": 125000}}`

When you compile the project, please make sure you change the board type to match yours (project is using by default a dev board from 4 years ago, and I know there has been many boards released in the meantime).

@obones  do you know why/if 250Kbit/s is the default for this radio chip and if it's supported and tests by you?

obones, 2023-05-23:
Only those values are allowed:
[image]

So 250 is indeed too far away from either 232 and 270 and RadioLib makes no effort to round the value to the nearest allowed one.
For information, I'm using 325 and 541 depending on what I'm listening to.

cpainchaud, 2023-05-23:
@obones  Would it make sense to force specific exact values (nearest match) in our code or submit a patch to RadioLib to make it happen ?

EBME2, 2023-05-23:
Thanks both I’ll have a play setting via the CLI and see what I get. Could this be set from the Web UI by a drop down with only the specific values in the table, though I guess these are different depending on the radio type? Just a thought. Appreciate you coming back to me.

EBME2, 2023-05-23:
Not really sure it is doing anything @cpainchaud you mentioned board type, unfortunately I'm using a cheap no name Chinese board but it does say 'DEVKITV1' on the bottom and using the 'esp32dev' to compile. I am getting the following when the radio initializes which I guess isn't right?

`Message arrived [Serial]:10;config;set;{"radio":{"rx_bandwidth": 541000}}

Saving JSON config to FLASH.... [  9184][E][vfs_api.cpp:105] open(): /littlefs/tmp.json does not exist, no permits for creation
OK
Start of async Receiver was requested but it's not enabled!
Config file saved to flash.
E (9311) gpio: gpio_isr_handler_remove(480): GPIO isr service is not installed, call gpio_install_isr_service() first
E (9318) gpio: gpio_wakeup_disable(579): GPIO number error
E (9323) gpio: gpio_set_intr_type(147): GPIO number error
Now trying to initialize hardware 'CC1101'
Initialized CC1101(freq=433.92Mhz,br=9.600kbps,rxbw=541.0khz)=0
CC1101 SetOOK(true)=0
CC1101 setPromiscuousMode(true)=0
CC1101 disableSyncWordFiltering(true)=0
Hardware initialization was successful!
E (9504) gpio: gpio_isr_handler_remove(480): GPIO isr service is not installed, call gpio_install_isr_service() first
E (9510) gpio: gpio_wakeup_disable(579): GPIO number error
E (9516) gpio: gpio_set_intr_type(147): GPIO number error
` 
And what I can't quite understand is the code seems to mention a pin RX_NA which isn't on pins to connect

` bool initialize_CC1101() {

      radioLibModule = Module(pins::RX_CS, -1, pins::RX_RESET, pins::RX_NA);
      if(radio_CC1101 == nullptr)
        radio_CC1101 = new CC1101(&radioLibModule);
      else
        *radio_CC1101 = &radioLibModule;
`
Again thanks for your help

EBME2, 2023-05-31:
OK I have made a little progress, I have found by enabling and then immediately disabling async_mode using the following commands it now starts to receive signals.

10;config;set;{"signal":{"async_mode_enabled":true}}
10;config;set;{"signal":{"async_mode_enabled":false}}

This needs to be done after each reboot, I'll do a bit more digging.

cpainchaud, 2023-05-31:
Interesting. Can you please confirm your ESP32 model ? written on the shield ....

cpainchaud, 2023-05-31:
what puzzles me is that message:
E (9504) gpio: gpio_isr_handler_remove(480): GPIO isr service is not installed, call gpio_install_isr_service() first
E (9510) gpio: gpio_wakeup_disable(579): GPIO number error
E (9516) gpio: gpio_set_intr_type(147): GPIO number error

It should appear only when you have async scanner enabled which is off by default

cpainchaud, 2023-05-31:
I don't see a reason to use RX_NA (actually I never seen any use RX_NA, it must be legacy stuff from legacy rflink project), RX_DATA is the receive pin you need.

cpainchaud, 2023-05-31:
you should reset your config entirely (10;config;reset;) or erase_flash to ensure that your unit will have no legacy configuration

obones, 2023-05-31:
> I don't see a reason to use RX_NA (actually I never seen any use RX_NA, it must be legacy stuff from legacy rflink project), RX_DATA is the receive pin you need.

That's a "work in progress" on my side that slipped through the PR for CC1101 support. But it should not matter because the default value for RX_NA is -1 which means "no pin".

EBME2, 2023-05-31:
I'll reset everything to default and carefully log all changes I make, start from the beginning and report back.

Thanks again for coming back to me, really appreciated .

EBME2, 2023-06-01:
A little bit more done after a clean rebuild and reset.

- It doesn't need RadioLib upgrading, the version here works
- I do need to issue the command "10;config;set;{"radio":{"rx_bandwidth": 135000}}'" this is a one of action
- Each reboot I need to issue the commands "10;config;set;{"signal":{"async_mode_enabled":true}}" and then "10;config;set;{"signal":{"async_mode_enabled":false}}"
- I needed to modify the firmware slightly and not really sure why more below

When I was doing some debugging I dumped the config of the CC1101 registers to the serial console just so as I can see what was going on, to do this I created a function that read the CC1101 registers, what I found was that unless I called this read of these registers no signals are received for some reason. Though I did read all the registers it seems to need to read about 6 for some reason, and not any specific ones from what I can tell. I have added the call of read of the registers here to 1_Radio.cpp as  debug_CC1101() 

```
    void set_Radio_mode_CC1101(States new_State, bool force)
    {
      // @TODO : review compatibility with ASYNC mode
      if (current_State != new_State || force)
      {
        switch (new_State)
        {
          case Radio_OFF: {
            ::detachInterrupt(digitalPinToInterrupt(pins::RX_NA));

            if( RFLink::Signal::params::async_mode_enabled )
              RFLink::Signal::AsyncSignalScanner::stopScanning();

            auto success = radio_CC1101->standby();
            if(success != 0 ){
              Serial.printf_P(PSTR("Failed to switch to standby mode (code=%i), we will try to reinitialize it later\r\n"), (int) success);
              hardwareProperlyInitialized = false;
            }
            break;
          }

          case Radio_RX: {

            auto success = radio_CC1101->receiveDirectAsync();
            if(success != 0 ) {
              Serial.printf_P(PSTR("ERROR: CC1101 receiveDirect()=%i, we will try to reinitialize it later\r\n"), (int) success);
              hardwareProperlyInitialized = false;
            }
 
            debug_CC1101();

            pinMode(pins::RX_DATA, INPUT);

```

The routine to print the registers is below, I haven't included them all for brevity. I have no explanantion why reading the registers suddenly brings things to life. I am now seeing the received signal count incrementing but it isn't decoding anything, not sure if it is just picking up background noise, have tried sending from one ESP to the other, I did try the CLI command  10;RFUDEBUG=ON; but I'm not seeing anything received on the CLI but the count on the WEB UI increments.

PS this is all very new to me in case you can't tell.

```
 void debug_CC1101() {  
      Serial.printf_P(PSTR("----- CC1101 Status -----\r\n"));
      Serial.printf_P(PSTR("CC1101_MDMCFG1: 0x%.2x\r\n"),
        radioLibModule.SPIreadRegister(RADIOLIB_CC1101_REG_MDMCFG1));
      Serial.printf_P(PSTR("CC1101_MDMCFG2: 0x%.2x\r\n"),
        radioLibModule.SPIreadRegister(RADIOLIB_CC1101_REG_MDMCFG2));
      Serial.printf_P(PSTR("CC1101_MDMCFG3: 0x%.2x\r\n"),
        radioLibModule.SPIreadRegister(RADIOLIB_CC1101_REG_MDMCFG3));
      Serial.printf_P(PSTR("CC1101_MDMCFG4: 0x%.2x\r\n"),
        radioLibModule.SPIreadRegister(RADIOLIB_CC1101_REG_MDMCFG4));
      Serial.printf_P(PSTR("-------------------------\r\n"));
      Serial.printf_P(PSTR("CC1101_DEVIATN: 0x%.2x\r\n"),
        radioLibModule.SPIreadRegister(RADIOLIB_CC1101_REG_DEVIATN));
      Serial.printf_P(PSTR("CC1101_AGCCTRL0: 0x%.2x\r\n"),
        radioLibModule.SPIreadRegister(RADIOLIB_CC1101_REG_AGCCTRL0));
      Serial.printf_P(PSTR("CC1101_AGCCTRL1: 0x%.2x\r\n"),
        radioLibModule.SPIreadRegister(RADIOLIB_CC1101_REG_AGCCTRL1));
```

obones, 2023-06-01:
What's the value of RX_NA ?

EBME2, 2023-06-01:
Sorry I forgot to say RX_NA pin assignment is the default value of -1 and GDO2 on CC1101 is not connected to any pin

EBME2, 2023-06-01:
Not sure what I have done but downloaded the latest firmware and can now send between the two ESP modules, so seems to be working, no need for RX_NA. Only thing I have to do is change the RX bandwidth to a supported one but otherwise successfully decoding data.

Thanks for all your help

closed 2023-06-01 by EBME2