cpainchaud/RFLink32 #82: Some Somfy RTS codes are not recognised

crazyserver, 2023-07-04:
When using the remote and checking the received MQTT message I have this code:

20;01;RTS;ID=684f3d;ID=01dd;CMD=UNKNOWN;

Checking with RTL 433 I get:

model 	: Somfy-RTS	id    	: 3D4F68
Control   : Up (6)   	Counter   : 245       	Retransmission: 0     	Integrity : CHECKSUM

After pairing the motors to RFLink, nothing happens for any command.

Checking at the RTL 433 code I can see that there's something named TEL FIX that makes the control button specification to differ.

https://github.com/merbanan/rtl_433/issues/1571

Could this be implemented on RFLink?

I'm not an expert on the topic, very newbie and the serial of my sonoff device is not working to get a proper debug.

crazyserver, 2023-07-04:
Not really related but can also help to improve RTS: https://github.com/merbanan/rtl_433/issues/2356

cpainchaud, 2023-07-06:
@obones would you mind having a look at this RTS code? thank you so much!

crazyserver, 2023-07-07:
I've been developing some chages you can find here: https://github.com/crazyserver/RFLink32/tree/rts-telfix When I finish the testing I will open a pull request.

obones, 2023-07-07:
Thanks for this.
Could you please be consistent in the bracing position style?
It's really weird to see those curly braces at the end of lines and `else` not on its line by itself when the rest of the code does it.

crazyserver, 2023-07-07:
Of course I'll do the changes when the code is finished!

closed 2023-10-11 by crazyserver