cpainchaud/RFLink32 #85: Add per device topics and json payload to MQTT

crazyserver, 2023-07-07:
Instead of just using one topic for send and one for receive, use a full topic tree based on protocol and device ID.

crazyserver, 2023-10-16:
Hi @cpainchaud,

I'd like to know if it would be great to implement that. I have something to start with that. I creates a topic for each device when a signal is received.

Example:
Topic: rflink/msg/RTS/1234ab
Payload: {code: '0000',cmd: 'DOWN'}

Instead of:
Topic: rflink/msg
Payload: 20;7F;RTS;ID=1234ab;CODE=0000;CMD=DOWN;

This way it uses standard mqtt model of passing data. My current version emits both the "plain" and the json approach and it's only used for received signals.

Pros: When using HomeAssistant, or any other MQTT broker you can listen to topics by device and then read the json having all the params separately. 

We can also include some RFLink code in the json.

Cheers,

Pau

cpainchaud, 2023-11-08:
Hi @crazyserver,

 I don't have the time to work on it but feel free to submit a patch and I will review&merge it