cpainchaud/RFLink32 #87: Guru Meditation Error and then Reboot when sending datas with NewKaku protocol

WinZs, 2023-07-09:
Hi ! first of all, thanks for all the amazing work you did ! 
I need some help please.  
Receiving data  works like a charm but I have issues with emission : 
my ESP32 nodeMCU always crash when I try to send data with the NewKaku protocol (I have DIO stuffs) 
It seem's that the crash only occurs with NewKaku protocol ; I've tried some example from https://www.rflink.nl/protref.php  Sample data of transmitted RF packets with others protocols : no crash.  
 
Here is the serial output log with 5 different data sent.
NewKaku always crash the esp with Guru Mediatation error and then reboot : 

12:41:53.615 -> NTP synchronized: 2023-07-09 10:41:52
12:42:39.179 -> [2K Message arrived [Serial]:10;Kaku;01769f0a;2;ON
12:42:39.179 -> 20;01;CMD UNKNOWN;
12:43:20.029 -> [2K Message arrived [Serial]:10;DELTRONIC;001c33;
12:43:20.029 -> 20;02;CMD UNKNOWN;
12:43:37.909 -> [2K Message arrived [Serial]:10;HomeEasy;7900b100;3;ON
12:43:37.943 -> 20;03;CMD UNKNOWN;
12:43:53.175 -> [2K Message arrived [Serial]:10;Kaku;00004d;1;OFF;
12:43:53.578 -> 20;04;OK;
12:44:11.549 -> [2K Message arrived [Serial]:10;NewKaku;128ac4d;1;OFF;
12:44:11.850 -> Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 
12:44:11.850 -> 
12:44:11.850 -> Core  1 register dump:
12:44:11.850 -> PC      : 0x4008476c  PS      : 0x00060c35  A0      : 0x800847ae  A1      : 0x3ffb2110  
12:44:11.850 -> A2      : 0x3ff5f070  A3      : 0x10ddd41a  A4      : 0x00000017  A5      : 0x3ffb81e8  
12:44:11.850 -> A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x10ddd41a  A9      : 0x3ff5f078  
12:44:11.884 -> A10     : 0x00000001  A11     : 0x00060c23  A12     : 0x00060c20  A13     : 0x00000001  
12:44:11.884 -> A14     : 0x00000019  A15     : 0x00000001  SAR     : 0x00000009  EXCCAUSE: 0x00000006  
12:44:11.884 -> EXCVADDR: 0x00000000  LBEG    : 0x40084781  LEND    : 0x40084789  LCOUNT  : 0x00000027  
12:44:11.884 -> 
12:44:11.884 -> 
12:44:11.884 -> Backtrace: 0x40084769:0x3ffb2110 0x400847ab:0x3ffb2130 0x400f831c:0x3ffb2150 0x400dbeec:0x3ffb2170 0x400de5f1:0x3ffb21b0 0x400e391a:0x3ffb21e0 0x400e5278:0x3ffb2200 0x400dcae5:0x3ffb2220 0x400e4f46:0x3ffb2240 0x400e5467:0x3ffb2270 0x400f9075:0x3ffb2290
12:44:11.917 -> 
12:44:11.917 -> 
12:44:11.917 -> Core  0 register dump:
12:44:11.917 -> PC      : 0x401949ea  PS      : 0x00060e35  A0      : 0x801029c4  A1      : 0x3ffbca50  
12:44:11.917 -> A2      : 0x00000000  A3      : 0x80000001  A4      : 0x800858f2  A5      : 0x3ffbf68c  
12:44:11.950 -> A6      : 0x3ffbe5dc  A7      : 0x3ffc92ec  A8      : 0x8010249a  A9      : 0x3ffbca20  
12:44:11.950 -> A10     : 0x00000000  A11     : 0x80000001  A12     : 0x80090ddc  A13     : 0x3ffbc950  
12:44:11.950 -> A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000001d  EXCCAUSE: 0x00000006  
12:44:11.950 -> EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
12:44:11.950 -> 
12:44:11.950 -> 
12:44:11.950 -> Backtrace: 0x401949e7:0x3ffbca50 0x401029c1:0x3ffbca70 0x4008eea8:0x3ffbca90
12:44:11.984 -> 
12:44:11.984 -> 
12:44:11.984 -> 
12:44:11.984 -> 
12:44:11.984 -> ELF file SHA256: 2e886cce41b4d59b
12:44:11.984 -> 
12:44:11.984 -> E (16509) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
12:44:11.984 -> Rebooting...
12:44:11.984 -> ets Jun  8 2016 00:22:57
12:44:11.984 -> 
12:44:11.984 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
12:44:11.984 -> configsip: 0, SPIWP:0xee
12:44:11.984 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
12:44:12.018 -> mode:DIO, clock div:2
12:44:12.018 -> load:0x3fff0030,len:1184
12:44:12.018 -> load:0x40078000,len:13192
12:44:12.018 -> load:0x40080400,len:3028
12:44:12.018 -> entry 0x400805e4
12:44:12.520 -> E (522) esp_core_dump_flash: No core d⸮f⸮⸮ partition found!
12:44:12.520 -> E (522) esp_core_dump_flash: No core dump partition found!
12:44:12.788 -> 
12:44:12.788 -> Arduino IDE Version :	10812
12:44:12.788 -> ESP SDK version:	v4.4.4
12:44:12.788 -> Sketch File :		RFLink/RFLink.cpp
12:44:12.788 -> Compiled on :		Jul  1 2023 at 14:05:45
12:44:12.788 -> 20;00;RFLink_ESP;VER=5.1;BUILD=unknown;
12:44:12.788 -> Loading persistent filesystem... OK. File system usage: 12/320KB.
12:44:12.821 -> Now opening JSON config file '/config.json'
12:44:12.855 -> JSON file mem usage: 1875 / 4096
12:44:12.855 -> Now dumping JSON file content:

cpainchaud, 2023-07-10:
Hi,

 Did you compile it by yourself? or used an image from this repo (or another?)

thank you

WinZs, 2023-07-10:
Hi ! 
I 've used RFLink32-master.zip  from here and compiled with Plateformio IDE as recommanded in documentation. 
It was the first time I used Plateformio...  so it is more than possible that I've missed someting..?..  In that case sorry for inconvenience. I will try to use last nightly bin image and give feddback.
Thanks for all 
Greetings

WinZs, 2023-07-10:
Dear C. Painchaud,
I 've updated from web interface firmware via http url : https://github.com/cpainchaud/RFLink32/releases/download/nightly/esp32-firmware-OTA.bin 

But still error Guru Mediatation error when emmitting (serial or mqtt)

I have 2 standard Rf module Qiachip WL 101
Receiving no problem , Mqtt works fine too

Arduino IDE Version :   10812
ESP SDK version:        v4.4.4
Sketch File :           RFLink/RFLink.cpp
Compiled on :           Jul 10 2023 at 08:09:39
20;00;RFLink_ESP;VER=5.1;BUILD=20230710-080645-master;
Loading persistent filesystem... OK. File system usage: 12/320KB.
Now opening JSON config file '/config.json'
JSON file mem usage: 1875 / 4096
Now dumping JSON file content:
{"wifi":{"client_enabled":true,"client_dhcp_enabled":true,"client_ssid":"XXXXXXXXXXXXXXX","client_password":"XXXXXXXXXXXXXXX","client_ip":"XXXXXXXXXXXXXXX","client_mask":"255.255.255.0","client_gateway":"XXXXXXXXXXXXXXX","client_dns":"XXXXXXXXXXXXXXX","client_hostname":"XXXXXXXXXXXXXXX","ap_enabled":true,"ap_ssid":"RFLink-AP","ap_password":"","ap_ip":"192.168.4.1","ap_network":"192.168.4.0","ap_mask":"255.255.255.0"},"mqtt":{"enabled":true,"server":"XXXXXXXXXXXXXXX","port":1883,"id":"RFLink32","user":"XXXXXXXXXXXXXXX","password":"XXXXXXXXXXXXXXX","topic_in":"/ESP00/cmd","topic_out":"/ESP00/msg","lwt_enabled":true,"topic_lwt":"/ESP00/lwt","ssl_enabled":false,"ssl_insecure":true,"ca_cert":""},"serial2net":{"enabled":false,"port":1900},"portal":{"enabled":true,"auth_enabled":true,"auth_user":"XXXXXXXXXXXXXXX","auth_password":"XXXXXXXXXXXXXXX"},"signal":{"async_mode_enabled":false,"sample_rate":1,"min_raw_pulses":24,"seek_timeout":25,"min_preamble":100,"min_pulse_len":90,"signal_end_timeout":5000,"signal_repeat_time":250,"scan_high_time":50},"radio":{"hardware":"generic","rx_data":21,"rx_vcc":-1,"rx_nmos":-1,"rx_pmos":-1,"rx_gnd":-1,"rx_na":-1,"rx_reset":-1,"rx_cs":-1,"tx_data":23,"tx_vcc":-1,"tx_nmos":-1,"tx_pmos":-1,"tx_gnd":-1}}
Now trying to initialize hardware 'generic'
Hardware initialization was successful!
Applied slicer 'Legacy'
Signal parameters have changed.
* WIFI AP starting with SSID 'RFLink-AP' and band=9... OK
Trying to connect WIFI SSID XXXXXXXXXXXXXXX. A status will be given whenever it occurs.
Radio pin RF_RX_DATA :  21
Radio pin RF_TX_DATA :  23
Starting WebServer... OK
Connected to AP SSID:XXXXXXXXXXXXXXX  BSSID: XXXXXXXXXXXXXXX  Channel: 11  Auth mode: 3
WiFi Client has received a new IP: XXXXXXXXXXXXXXX
MQTT parameters have changed, first disconnecting...
MQTT parameters have changed, now applying...
Trying to connect to MQTT Server 'XXXXXXXXXXXXXXX' ... Established
MQTT ID :               RFLink32
MQTT Username :         XXXXXXXXXXXXXXX

NTP synchronized: 2023-07-10 09:12:29

20;01;NewKaku;ID=XXXXXXXXXXXXXXX;SWITCH=1;CMD=ON;
20;02;NewKaku;ID=XXXXXXXXXXXXXXX;SWITCH=2;CMD=OFF;
20;03;NewKaku;ID=XXXXXXXXXXXXXXX;SWITCH=2;CMD=ON;

Message arrived [MQTT] 10;NewKaku;XXXXXXXXXXXXXXX;2;ON;
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Core  1 register dump:
PC      : 0x400f83f0  PS      : 0x00060f35  A0      : 0x800dbf3f  A1      : 0x3ffb2070
A2      : 0x00000000  A3      : 0x04719bf3  A4      : 0x3ffd397c  A5      : 0x3ffb81e8  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x800f83cf  A9      : 0x3ffb2050  
A10     : 0x047198f1  A11     : 0x00000000  A12     : 0x047196df  A13     : 0x00000001  
A14     : 0x3ffc712c  A15     : 0x3ffc7144  SAR     : 0x00000009  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x40084781  LEND    : 0x40084789  LCOUNT  : 0x00000026  

Backtrace: 0x400f83ed:0x3ffb2070 0x400dbf3c:0x3ffb2090 0x400de641:0x3ffb20d0 0x400e396a:0x3ffb2100 0x400e52cc:0x3ffb2120 0x400dc9ea:0x3ffb2140 0x400e3d37:0x3ffb2160 0x40192ee7:0x3ffb2180 0x400e5c71:0x3ffb21a0 0x400e468a:0x3ffb21e0 0x400e4f8b:0x3ffb2240 0x400e54bb:0x3ffb2270 0x400f9125:0x3ffb2290

  #0  0x400f83ed:0x3ffb2070 in initArduino at C:/Users/XXXXXXXXXXXXXXX/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-misc.c:220
  #1  0x400dbf3c:0x3ffb2090 in RFLink::Signal::AC_Send(unsigned long, unsigned char) at RFLink/2_Signal.cpp:910
  #2  0x400de641:0x3ffb20d0 in Plugin_016(unsigned char, char const*) at RFLink/./Plugins/Plugin_016.c:37
  #3  0x400e396a:0x3ffb2100 in saveRTSRecord(unsigned char, unsigned int, unsigned short) at RFLink/./Plugins/Plugin_017.c:554
  #4  0x400e52cc:0x3ffb2120 in ArduinoJson6185_91::VariantSlot::next() at .pio/libdeps/genericESP32/ArduinoJson/src/ArduinoJson/Variant/VariantSlot.hpp:42
      (inlined by) ArduinoJson6185_91::VariantSlot* ArduinoJson6185_91::CollectionData::getSlot<ArduinoJson6185_91::StringAdapter<char const*, false, void> >(ArduinoJson6185_91::StringAdapter<char const*, false, void>) const at .pio/libdeps/genericESP32/ArduinoJson/src/ArduinoJson/Collection/CollectionImpl.hpp:112
      (inlined by) ArduinoJson6185_91::VariantData* ArduinoJson6185_91::CollectionData::getOrAddMember<ArduinoJson6185_91::StringAdapter<char const*, false, void> >(ArduinoJson6185_91::StringAdapter<char const*, false, void>, ArduinoJson6185_91::MemoryPool*) at .pio/libdeps/genericESP32/ArduinoJson/src/ArduinoJson/Collection/CollectionImpl.hpp:148
      (inlined by) ArduinoJson6185_91::VariantData* ArduinoJson6185_91::objectGetOrAddMember<ArduinoJson6185_91::StringAdapter<char const*, false, void> 
[... cut ...]
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
E (522) esp_core_dump_flash: No core du���partition found!
E (522) esp_core_dump_flash: No core dump partition found!

Arduino IDE Version :   10812
ESP SDK version:        v4.4.4
Sketch File :           RFLink/RFLink.cpp
Compiled on :           Jul 10 2023 at 08:09:39
20;00;RFLink_ESP;VER=5.1;BUILD=20230710-080645-master;
Loading persistent filesystem... OK. File system usage: 12/320KB.
[...]

closed 2026-11-13 by cpainchaud