cpainchaud/RFLink32 #93: Help creating entities from MQTT

JacquesdeBruyn, 2023-10-16:
Hi, 

Probably a stupid question but for the life of me I can't figure out how to use the mqtt output. I've never used mqtt before and can't seem to find a sample setup that I can understand.

I'm trying to setup multiple PIR sensors. The first one gives me this code in mqtt:
20;AF;EV1527;ID=0daed6;SWITCH=09;CMD=ON;

others send a code when triggered and when trigger is over(LED turning on and off)
20;BF;EV1527;ID=01d108;SWITCH=09;CMD=ON;
and
20;C0;EV1527;ID=01d108;SWITCH=00;CMD=ON;

I'm not bothered about the switch off. Assuming I need to bind a trigger to message received from "ID"

Any help on how to setup one of these would be appreciated.

JacquesdeBruyn, 2023-10-16:
Which helper do you suggest I use for this? I have used helpers to average battery readings but never anything to use mqtt?

JacquesdeBruyn, 2023-10-16:
I have automation setup, how do I deal with the first bit changing?
20;AF;EV1527;ID=0daed6;SWITCH=09;CMD=ON;
The 20:AF bit changes every time?

JacquesdeBruyn, 2023-10-16:
Automation etc working, Just need a way to split different sensors apart, I have these three for now. 

20;C8;EV1527;ID=0b5be2;SWITCH=09;CMD=ON;

20;C9;EV1527;ID=01d108;SWITCH=09;CMD=ON;

20;CA;EV1527;ID=0daed6;SWITCH=09;CMD=ON;

Can I filter out the Message counter?

crazyserver, 2023-10-17:
The ID is the device Id and the first 20;XX should be something like a checksum? So I'd have different automations or different triggers for every line here.

If I manage to get the Json solution integrated it will be easier to manage this.

I think I cannot help you more than that. Try to figure out all the possibilities, you can also skip the payload check and add a condition:

```
alias: MQTT listener example without payload filter
description: ""
trigger:
  - platform: mqtt
    topic: /rflink/msg
condition:
  - condition: template
    value_template: >-
      {{ trigger.payload == "20;C8;EV1527;ID=0b5be2;SWITCH=09;CMD=ON;" or
      trigger.payload == "20;C8;EV1527;ID=0b5be2;SWITCH=09;CMD=OFF;" }}
    enabled: true
action:
  - if:
      - condition: template
        value_template: >- {{ trigger.payload == "20;C8;EV1527;ID=0b5be2;SWITCH=09;CMD=ON;" }}
        enabled: true
    then:
      - service: input_boolean.turn_on
        data: {}
        target:
          entity_id: input_boolean.test
    else:
      - service: input_boolean.turn_off
        data: {}
        target:
          entity_id: input_boolean.test
mode: single

```

I don't know if the codes are more complex, try using regular expression on the conditions.

Hope this helps

JacquesdeBruyn, 2023-10-17:
I have it somewhat working, but I can't split devices apart so it just responds to all of them.

The 20;XX isn't a checksum. It's a counter. Counts up one for every message received. I was hoping that there's a way to ignore that bit like you'd do in CMD on pc by replacing characters with an *

JacquesdeBruyn, 2023-10-17:
For clarity, I have 3 wireless PIRs at the moment, going to add more so would like to eventually be able to know which one triggered. As these are going to be moved around as a sort of cheap security against theft in our vegetable garden.

crazyserver, 2023-10-17:
So I suggest you to investigate with template conditions and jinja templates, using regular expressions it's your solution.