cpainchaud/RFLink32 #103: cc1101 bandwith setting

pwwoj, 2023-12-14:
I've compiled project for esp32/cc1101 with latest radio library and it seems that rxBandwidth setting of 250000 does not pass validity checks. Below clacs from spec , so it looks like some container with predefined values might be used - I have succesfully used defualt 203000 or 270000 as well.
[image]

andyboeh, 2024-01-21:
I just ran into the same issue. Looks like the config option for the Rx bandwidth is already there but it is not (yet) supported by the web interface. Changing the default to 270000 initializes the module properly.

DanielKinsman, 2025-01-04:
I just ran into this as well. I'm not sure how to set the config at all apart from editing the code? I used the value 232.0 from here: https://github.com/jgromes/RadioLib/blob/master/src/modules/CC1101/CC1101.cpp#L439