cpainchaud/RFLink32 #108: Help to send RTS commands

vaguiner7, 2024-01-26:
Hi there, I am using a Sonoff RF BridgeR2 that I did the necessary changes following the [guide](https://github.com/cpainchaud/RFLink32/blob/master/doc/sonoff_rf_bridge.md)

My issue is, I can decode the commands sent from my [SCE-R1](https://www.tube-motors.com/worldwide/Item/4/165) control but I cannot send it from the Sonoff Bridge.

The tests:

01 - Sending the command STOP from my SCE-R1 control I got the results (worked):

- Serial port on the Sonoff

[image]

- [433_rtl](https://github.com/merbanan/rtl_433) with [820T2 SDR](https://pt.aliexpress.com/item/1005006467185080.html)

[image]

02 - So I used the examples on the [RTS plugin code](https://github.com/cpainchaud/RFLink32/blob/master/RFLink/Plugins/Plugin_017.c) to register a new control on the Sonoff.

[image]

And I tried to send an UP command, but could not see the command on 433_rtl like when I sent from the SCE-R1 control.

[image]

To make sure that some signal was  sent I used the [URH ](https://github.com/jopohl/urh)tool to see the spectrum, and I got something.

[image]

My conclusions so far is, the sonoff can send the signal, but the signal is not recognized as a Sonfy-RTS signal.

Do you guys have an idea that can help me? I am not sure if what I did to configure was correct.

Thank you

vaidas911, 2024-11-18:
Somfy uses different frequency. Use CC1101 and some esp32 and https://github.com/rstrouse/ESPSomfy-RTS

vaguiner7, 2024-12-04:
It worked pretty well, thanks a lot @vaidas911

closed 2024-12-04 by vaguiner7