cpainchaud/RFLink32 #109: Incorrect "SWITCH" value when sending

TomiBp, 2024-01-31:
I have quite a few Telldus / Proove EWT0006 type remote controls 
and I have two Sonoff RF Bridge R2 V2.2 

This is the code for the remote control buttons:

1 ON
`20;4D;NewKaku;ID=02594ff2;SWITCH=10;CMD=ON;`
2 ON
`20;4E;NewKaku;ID=02594ff2;SWITCH=f;CMD=ON;`
3 ON
`20;4F;NewKaku;ID=02594ff2;SWITCH=e;CMD=ON;`
ALL ON
`20;50;NewKaku;ID=02594ff2;SWITCH=10;CMD=ALLON;`

1 OFF
`20;4B;NewKaku;ID=02594ff2;SWITCH=10;CMD=OFF;`
2 OFF
`20;4C;NewKaku;ID=02594ff2;SWITCH=f;CMD=OFF;`
3 OFF
`20;4D;NewKaku;ID=02594ff2;SWITCH=e;CMD=OFF;`
ALL OFF
`20;4E;NewKaku;ID=02594ff2;SWITCH=f;CMD=ALLOFF;`

I realized that there is an error in the program code.
If I want to clone the remote controls code, 
I have to decrease the value of "SWITCH" by one. 
For example, if I want to send a `1`, I have to send a `0`, 
or if I want to send an `F`, I have to send an `E`.

My problem is that if I want to send a `10` and I send a `9`, 
it doesn't send a `10`, but an `A`.

For this reason, I cannot clone the "1 ON", "1 OFF" and "ALL ON" buttons of the remote control.

If I send this line, 
`10;NewKaku;004adff2;9;ON;`
the sending is successful,
`20;57;OK;`
but the sent code is wrong
`20;97;NewKaku;ID=004adff2;SWITCH=a;CMD=ON;`

**What code should I send so that the sent code is `SWITCH=10`, not `SWITCH=a`?**

[image]
[image]

TomiBp, 2024-02-01:
**I found the answer to my own question.**
I also wrote a NodeRed function bubble called 
"RfLink - Correction of incorrect `SWITCH` value".
If you are also using NodeRed to send MQTT messages 
to RfLink, this may be useful for you as well.

```
msg.payload = msg.payload.replace(";1;", ";0;");
msg.payload = msg.payload.replace(";2;", ";1;");
msg.payload = msg.payload.replace(";3;", ";2;");
msg.payload = msg.payload.replace(";4;", ";3;");
msg.payload = msg.payload.replace(";5;", ";4;");
msg.payload = msg.payload.replace(";6;", ";5;");
msg.payload = msg.payload.replace(";7;", ";6;");
msg.payload = msg.payload.replace(";8;", ";7;");
msg.payload = msg.payload.replace(";9;", ";8;");
msg.payload = msg.payload.replace(";a;", ";9;");
msg.payload = msg.payload.replace(";b;", ";a;");
msg.payload = msg.payload.replace(";c;", ";b;");
msg.payload = msg.payload.replace(";d;", ";c;");
msg.payload = msg.payload.replace(";e;", ";d;");
msg.payload = msg.payload.replace(";f;", ";e;");
msg.payload = msg.payload.replace(";10;", ";f;");

return msg;
```

liangli2718, 2024-08-13:
#71