cpainchaud/RFLink32 #122: Is there a way to get an RA-01 without a DIO2 pin work?

DanielKinsman, 2024-12-15:
I stupidly bought this module which does not expose the DIO2 pin of the SX1278: https://www.aliexpress.com/item/1005006864170059.html?spm=a2g0o.order_list.order_list_main.15.2f1e1802phqlwc

[image]

Apart from attempting to disassemble and re-solder the chip, is there anything I can do in software to get it to work with the DIO0 pin?

Thanks for making this great software.

cpainchaud, 2024-12-16:
may be directly on the module?

[image]

[image]

DanielKinsman, 2024-12-16:
I tried but sadly it did not work, probably due to my poor soldering skills.

[image]

Netcat just gives:
`Now trying to initialize hardware 'SX1278'
Hardware failed to initialize, we will retry later!`

Thanks for your help though.

closed 2024-12-16 by DanielKinsman